Livox-SDK/livox_horizon_loam #2: Run times error

sczgzzzj, 2020-02-19:
[ERROR] [1582104656.932088632]: Client [/livox_repub] wants topic /livox/lidar to have datatype/md5sum [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6], but our version has [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181]. Dropping connection.

What's the problem, please

mbennehar, 2020-02-20:
They use the custom cloud format, in your driver launch you need to change xfer_format from 0 to 1

sczgzzzj, 2020-02-21:
Thank you, it's ready to use, but the test results are not so good

mbennehar, 2020-02-21:
Try this one instead: https://github.com/hku-mars/loam_livox

PanZhichen, 2020-02-25:
hi @sczgzzzj , if you want to run the demo directly, you need to launch the livox_ros_driver with livox_lidar_msg.launch to allow CustomMsg output. Or you can run the rosbag example, the[ download link](https://terra-1-g.djicdn.com/65c028cd298f4669a7f0e40e50ba1131/demo/parking_lot.bag) was updated recently.

Livox-SDK, 2020-02-25:
Hi,sczgzzzj and mbennehar
    The livox_horizon_loam only support the CustomMsg format, must use livox_lidar_msg.launch to launch the livx_ros_driver.

sczgzzzj, 2020-02-25:
tanks，I'll test it to see if it works

closed 2020-05-06 by PanZhichen