Livox-SDK/livox_horizon_loam #5: Point cloud export

iseli-c, 2020-02-25:
I have successfully run the loam processing on the sample rosbag files. How can the oriented results be exported to a point cloud file (such as .pcd?)

PanZhichen, 2020-05-28:
you can subscribe the topic "/velodyne_cloud_registered" and register them together to generate a pointclod map

cfessler, 2020-06-11:
hello, i am sorry to ask here but i am totally lost now. i got it to run and also play the rosbag file. But now i do not know how to proceed. Is there any guide or hint how to get to a point cloud file (such as .las?)

tranminhduc4796, 2021-04-02:
I hope my solution is still soon enough for everyone who've been struggle with this issue.
Belows are what I did:

1. Use package_assembler to stack PointCloud2 messages into one from topic /velodyne_cloud_registered or /laser_cloud_surround (I recommend the first one) and publish the message to a topic.
2. Develop a subscriber to listen to the topic you published the message.
The step 1 and 2 you can follow this [guide](https://www.theconstructsim.com/merge-laser-scans-single-pointcloud/).
3. Use package pointcloud_to_pcd to export the message from the topic you defined to .pcd file.
4. (Optional) In case your map is too large, you can turn on/off your subscriber to generate multiple pcd files. And use pcl_concatenate_points_pcd to concat those .pcd files.
*Note*: I just realized that you can just use pointcloud_to_pcd to convert messages from /velodyne_cloud_registered or /laser_cloud_surround to .pcd files and then use pcl_concatenate_points_pcd to concat all of them. I haven't tested this way yet but I guess I would work well.