Livox-SDK/livox_horizon_loam #9: Uses up all the memory

griz1112, 2020-05-23:
Noticed that this code uses up all the memory on  the Xavier I'm running it on after a period of time? What is causing this? The computer has 16gig. Does it release this memory at some point or does it continue to need more?

PanZhichen, 2020-05-28:
If you use rviz for visualization, it will store all points to display the global map, which may cause memory consumption.

huangsiyuan0717, 2021-07-06:
@griz1112 @PanZhichen Hi! I met the same problem, my mem is 16g, I can hardly play a 18 mins data.Do you have a better way to solve this problem?

griz1112, 2021-07-06:
> @griz1112 @PanZhichen Hi! I met the same problem, my mem is 16g, I can hardly play a 18 mins data.Do you have a better way to solve this problem?

I don't use the mapping node just the node that creates the pointcloud. Combine that with the output from a zed camera into rtabmap_ros for mapping and the pointcloud from the livox for ros cost maps. Rtabmap uses camera input only for loop closure. The pointcloud node I use is from loam-livox. It keeps everything in memory unlike rtabmap which uses a disk based database to store loop closure info so it doesn't eat up all the memory.

huangsiyuan0717, 2021-07-13:
> I don't use the mapping node just the node that creates the pointcloud. Combine that with the output from a zed camera into rtabmap_ros for mapping and the pointcloud from the livox for ros cost [...]

Hi! Thanks for your great idea.
Can you provide some reference materials for your code, I am a new students in slam, just don't how to achieve in code.

griz1112, 2021-07-13:
> Hi! Thanks for your great idea.
> Can you provide some reference materials for your code, I am a new students in slam, just don't how to achieve in code.

You need to grab this package https://github.com/hku-mars/loam_livox/tree/master/ Then edit livox.launch in the launch dir. Remove or comment out this line. <node pkg="loam_livox" type="livox_laserMapping" name="livox_laserMapping" output="screen" /> that stops it from mapping and just gives you the pointcloud. Use that file to launch the package and you will get the pointcloud on the /livox/lidar topic

huangsiyuan0717, 2021-07-13:
Thanks for your sharing!

huangsiyuan0717, 2021-07-13:
> Thanks for your sharing!

> > > > @griz1112 @PanZhichen Hi! I met the same problem, my mem is 16g, I can hardly play a 18 mins data.Do you have a better way to solve this problem?
> > > 
> > > 
> > > I don't use the mapping node just the node that creates the pointcloud. Combine that with the output from a zed camera into rtabmap_ros for mapping and the pointcloud from the livox for ros cost maps. Rtabmap uses camera input only for loop closure. The pointcloud node I use is from loam-livox. It keeps everything in memory unlike rtabmap which uses a disk based database to store loop closure info so it doesn't eat up all the memory.
> > 
> > 
> > Hi! Thanks for your great idea.
> > Can you provide some reference materials for your code, I am a new students in slam, just don't how to achieve in code.
> 
> You need to grab this package https://github.com/hku-mars/loam_livox/tree/master/ Then edit livox.launch in the launch dir. Remove or comment out this line loam_livox" type="livox_laserMapping" name="livox_laserMapping" output="screen. That stops it from mapping and just gives you the pointcloud. Use that file to launch the package and you will get the pointcloud on the /livox/lidar topic

Thanks for your sharing!

griz1112, 2021-07-13:
Its not my code.