Livox-SDK/livox_horizon_loam #12: running and generating a pointcloud (.las)

cfessler, 2020-06-12:
i successfully installed the loam following the guide, but now i am stuck a bit. Can you guide me how to generate a pointcloud out of that. 

as far as i have understood i have to choose the ros every time i open the console:
` source ~/catkin_ws/devel/setup.bash`

then i run
`roslaunch loam_horizon loam_livox_horizon_imu.launch`
to get the data from lidar and IMU?

whilst
`rosbag play YOUR_DOWNLOADED.bag` - in my case `rosbag play 2020_open_road.bag`
will add the lidar data to the file?
i do not really get it and to be honest i did not unterstand the ros documentation.

So this is the point where i am standing, is there someone able and willing to point me into the right direction?
When looking for the points mentioned i found the line 
`rosrun pcl_ros pointcloud_to_pcd input:=/velodyne/pointcloud2`
which looks like refering to the subscibe to `/velodyne_cloud_registered` and register, but i dont get it and i dont really get how the magic works. And what ist meant by registering them together?

Anyone who is able to help me with that?

telemething, 2020-06-13:
Are you familiar enough with ROS to know that it publishes ‘messages’ on ‘topics’?

- Mark West

cfessler, 2020-06-13:
> Are you familiar enough with ROS to know that it publishes ‘messages’ on ‘topics’? - Mark West 
thank you for your reply and trying to point me in the right direction,
more or less but i would not say familiar; in my own words: ROS publishes what i gets from lidar via messages and if i understand the graph from `rqt_graph` right there happens some magic.
BUT what i do not understand is: it looks to like it is serial data (messages) so it must be somehow processed to get a map (or better pointloud).
NEW (didnt work the times i tried that before) when running
`rosrun pcl_ros pointcloud_to_pcd input:=/velodyne_cloud_registered` it now saves .pcd-files
please see rosgraph, is it right this way?
[image]

I am not so sure now, how can i proceed with those files? is generating the pcd-files the right way to come to a pointcloud? what should be my next steps?