Livox-SDK/livox_horizon_loam #14: [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!

nbelanger99, 2020-07-03:
I'm running the livox horizon lidar on one PC and running the loam on a different PC. 
After about 20 seconds of running LOAM, the laserOdometry module crashes with 

> [laserOdometry-2] process has died [pid 6983, exit code -11, cmd /home/abacus/catkin_ws/devel/lib/loam_horizon/laserOdometry __name:=laserOdometry __log:=/home/abacus/.ros/log/7a965bc2-bd63-11ea-8c9c-dca632b01010/laserOdometry-2.log].
> log file: /home/abacus/.ros/log/7a965bc2-bd63-11ea-8c9c-dca632b01010/laserOdometry-2*.log
> 
The `[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud! ` error is seen for a few seconds before the crash. 
The logs from the console is here: 
```

map prepare time 0.088757 ms
map corner num 2328  surf num 1720 
build tree time 0.470938 ms 
mapping data assosiation time 0.156549 ms 
mapping solver time 0.292138 ms 
Ceres Solver Report: Iterations: 8, Initial cost: 1.103842e-01, Final cost: 9.609673e-02, Termination: CONVERGENCE
mapping data assosiation time 0.148910 ms 
mapping solver time 0.166832 ms 
Ceres Solver Report: Iterations: 5, Initial cost: 8.602315e-02, Final cost: 8.225180e-02, Termination: CONVERGENCE
mapping optimization time 1.270840 
add points time 0.018058 ms
filter time 0.211446 ms 
mapping pub time 0.733810 ms 
whole mapping time 2.343390 ms +++++
[ INFO] [1593805376.805134363]: integrate rotation angle [x, y, z]: [1.94, -0.33, -3.99]
points size 22824 
prepare time 3.416602 
sort q time 4.660179 
seperate points time 8.831633 
scan registration time 18.150661 ms *************
data association time 14.873315 ms 
solver time 15.560551 ms 
data association time 6.526456 ms 
solver time 5.991541 ms 
optimization twice time 43.406445 
t_w_curr:   -6.6868   3.92787 0.0714668
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
publication time 5.321362 ms 
whole laserOdometry time 48.761081 ms 
 
map prepare time 0.096470 ms
map corner num 2329  surf num 1720 
build tree time 0.520440 ms 
mapping data assosiation time 0.173800 ms 
mapping solver time 0.275571 ms 
Ceres Solver Report: Iterations: 6, Initial cost: 9.436523e-02, Final cost: 6.917717e-02, Termination: CONVERGENCE
mapping data assosiation time 0.164770 ms 
mapping solver time 0.185270 ms 
Ceres Solver Report: Iterations: 5, Initial cost: 5.967188e-02, Final cost: 5.744594e-02, Termination: CONVERGENCE
mapping optimization time 1.359949 
add points time 0.010248 ms
filter time 0.241848 ms 
mapping pub time 0.814104 ms 
whole mapping time 2.547101 ms +++++
[ INFO] [1593805376.902820965]: integrate rotation angle [x, y, z]: [0.69, -0.35, -3.59]
points size 22808 
prepare time 1.079317 
sort q time 1.241579 
seperate points time 2.031818 
scan registration time 5.068313 ms *************
data association time 3.128026 ms 
solver time 4.016765 ms 
data association time 3.586069 ms 
solver time 2.727348 ms 
optimization twice time 13.746803 
t_w_curr:  -6.72343   4.02965 0.0620396
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
publication time 3.788966 ms 
whole laserOdometry time 17.590642 ms 
 
[... 1006 lines cut ...]
[ INFO] [1593805381.129251967]: integrate rotation angle [x, y, z]: [0.50, -1.42, 0.61]
points size 22799 
prepare time 3.339527 
sort q time 4.510507 
seperate points time 7.281703 
scan registration time 13.092277 ms *************
[ INFO] [1593805381.231206212]: integrate rotation angle [x, y, z]: [1.51, 0.28, 0.36]
points size 22803 
prepare time 4.910754 
sort q time 4.589096 
seperate points time 7.680346 
scan registration time 17.230554 ms *************
^C[imu_process-5] killing on exit
[rviz-6] killing on exit
[livox_repub-4] killing on exit
[ WARN] [1593805381.292699634]: catch sig 2
[ INFO] [1593805381.292836625]: b_exit=true, exit
[ INFO] [1593805381.292933963]: Wait for process loop exit
[laserMapping-3] killing on exit
[scanRegistration-1] killing on exit
terminate called without an active exception
shutting down processing monitor...
... shutting down processing monitor complete
done
```

echoGee, 2020-07-08:
Any updates on this ?

echoGee, 2020-07-10:
Could anyone give some pointers on debugging this ?

PanZhichen, 2020-07-31:
If you have this crash while playing a custom rosbag, please check if there is an empty point cloud frame in the rosbag.

echoGee, 2020-07-31:
We have had this from a live device being plugging in to the system too.

altfool, 2022-03-04:
Use ubuntu 18 system!

I encountered this problem as well. At that time, I used ubuntu 20 system. And I can't even start the launch file well. After I switched to ubuntu 18 with a full desktop version of ros-melodic which automatically includes the eigen3 and pcl. I only need to install the ceres library and it works!!!