Livox-SDK/livox_horizon_loam #16: [ERROR] [1594718132.933929306]: clear lidar buffer, only happen at the beginning

jackychung, 2020-07-14:
I create a node of external IMU instead of internal IMU on horizon.
And i use launch file livox_lidar_msg.launch  loam_horizon loam_livox_horizon_ext_imu.launch(include external IMU node).
But i get the following errors:

[ERROR] [1594718132.433931164]: clear lidar buffer, only happen at the beginning
[ERROR] [1594718132.533930171]: clear lidar buffer, only happen at the beginning
[ERROR] [1594718132.633928597]: clear lidar buffer, only happen at the beginning
[ERROR] [1594718132.733928756]: clear lidar buffer, only happen at the beginning
[ERROR] [1594718132.833929117]: clear lidar buffer, only happen at the beginning
[ERROR] [1594718132.933929306]: clear lidar buffer, only happen at the beginning
[ERROR] [1594718133.033931449]: clear lidar buffer, only happen at the beginning
[ERROR] [1594718133.133926509]: clear lidar buffer, only happen at the beginning
[ERROR] [1594718133.233926980]: clear lidar buffer, only happen at the beginning
[ERROR] [1594718133.333933161]: clear lidar buffer, only happen at the beginning

jackychung, 2020-07-14:
How to use external IMU ?

PanZhichen, 2020-07-17:
Please check if IMU and Lidar have the same time source?