Livox-SDK/livox_horizon_loam #17: How can i use the Livox Horizon Lidar to mapping AND too few points in my rviz

chewing-void, 2020-07-29:
Now I have a Livox Horizon Lidar,and I can use the Lidar to displays the point cloud image of the current shot, but I don't understand how to mapping  a omnibearing map. 
Second, when I played the rosbag(parking or road) ,the amount of point is so small. different from the picture in readme

chewing-void, 2020-07-30:
I found that whether i run bag or mapping, i can only display the first screen and will not be updated after that. In the terminal which I input roslaunch loam_horizon loam_livox_horizon.launch , it will display 

[laserMapping-4]_ process has died [pid 12705, exit code -11, cmd /home/void/catkin_ws/devel/lib/loam_horizon/laserMapping __name:=laserMapping __log:=/home/void/.ros/log/3ee1f118-d208-11ea-bfd5-f8e4e303098a/laserMapping-4.log].
log file: /home/void/.ros/log/3ee1f118-d208-11ea-bfd5-f8e4e303098a/laserMapping-4*.log

SherlockHermes, 2020-07-31:
I have the problem, how do you solve it? Thanks.

PanZhichen, 2020-07-31:
hi @chewing-void ~This is because the laserMapping node is dead, so the point cloud will not accumulate. Did you start roslaunch in ubuntu 16.04?

SherlockHermes, 2020-07-31:
> hi @chewing-void ~This is because the laserMapping node is dead, so the point cloud will not accumulate. Did you start roslaunch in ubuntu 16.04?

I run the code on ubuntu16.04. Other environments are as follow：
ceres:1.14.0, PCL:1.9.1, Eigen:3.3.7. Does PCL version cause this error？

PanZhichen, 2020-07-31:
@SherlockHermes This should be the key to this problem，you can test with PCL 1.7, good lunk!

chewing-void, 2020-08-01:
> hi @chewing-void ~This is because the laserMapping node is dead, so the point cloud will not accumulate. Did you start roslaunch in ubuntu 16.04?

Thanks for u help.I run the code on ubuntu 18.04 . Should I install 16.04 ?

SherlockHermes, 2020-08-01:
@PanZhichen u r right，when I change the PCL version to 1.7, the results get better. Thanks!

chewing-void, 2020-08-01:
@PanZhichen  When I change the PCL version to 1.7 , some error happened. It means that ubuntu 18.04 maybe not install PCL 1.7.

Olive-Wang, 2020-12-07:
I also have the similar issue, could anyone give a solution?

> [laserOdometry-2] process has died [pid 21819, exit code -11, cmd /home/nvidia/workspace/drone_ws/devel/lib/loam_horizon/laserOdometry __name:=laserOdometry __log:=/home/nvidia/.ros/log/c43c21ca-3870-11eb-b63f-00e02def12ea/laserOdometry-2.log].
log file: /home/nvidia/.ros/log/c43c21ca-3870-11eb-b63f-00e02def12ea/laserOdometry-2*.log