Livox-SDK/livox_horizon_loam #19: Error playing my rosbag file

Doublehes, 2020-09-13:
When I was playing my own rosbag file **recorded by livox_viewer**, i got an error:

[ERROR] [1599999355.900913770]: Client [/livox_scanRegistration] wants topic /livox/lidar to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6]. Dropping connection.

I do not know why it happened?
How can I solve this problem?

h-k8888, 2020-09-21:
I met the same problem
wrong pointcloud type
set the arg "output_type" to "1" in the file lvx_to_rosbag.launch
then convert .lvx to rosbag

Doublehes, 2020-11-07:
all right, solving by changing the output_type  in the file lvx_to_rosbag.launch

closed 2020-11-07 by Doublehes