Livox-SDK/livox_horizon_loam #21: The frame rate displayed on rviz drops rapidly

jacklidar, 2020-10-14:
Hi，We use Livox loam with ubuntu 16.04 + virtualbox to run Livox-Horizon-LOAM by command 'roslaunch loam_horizon loam_livox_horizon.launch', rviz started displaying at 30fps, and then start dropped slowly to 1fps. If you click reset, the frame rate will be restored to 30fps. We use rosbag to run with 'rosbag play 2020_parking_lot.bag', the issue also reproduce too. This problem we also email to dev@livoxtech.com, but we don't get any information about this  v _ v

How can I increase the frame rate of the display, because there is no problem when using livox_lidar_rviz.launch,  but the configuration file is not suitable for Livox-Horizon-LOAM. Is there any parameter settings for loam_livox_horizon.launch? How can we resolve this issue?

huangsiyuan0717, 2021-07-06:
I have the same problem, have you solve it?

huangsiyuan0717, 2021-07-06:
@jacklidar I have the same problem, have you solve it?