Livox-SDK/livox_horizon_loam #23: Scan Accuracy

iSIlya, 2020-11-20:
Hello. How to achieve the accuracy of the map so that the error is within 2 cm? Now the error as a result of LOAM is 10 cm. The scanner is used by LIVOX Horizont. Or is there another way to get a map with the lowest possible error?

VisionaryMind, 2021-01-03:
Livox Horizon and the latest Avia both can only be used outdoors. There is 10cm or greater inaccuracy in close proximity to walls. This is not information that Livox has published, and we have had to discover the problem the hard way. If you are moving in a straight line at less than 5km / hr, it can be used indoors, but you must avoid sharp panning, pitching and heading (turns).

In reality, these LiDARs are not capable of doing anything more than moderately accurate obstacle avoidance when applied to cars. They are not robust enough to handle LOAM except outdoors at very slow, careful speeds. We have mounted both the Horizon and the Avia to a drone, and the IMU has a great deal of drift plus the scan rate is not fast enough to capture anything when the drone is in motion (too much blur).

amirrezasadeqi, 2023-01-22:
Hello

Dear @VisionaryMind, Thanks for sharing your experience using Avia on a drone. I have a question about the blurry output you've mentioned.

Was the raw sensor output blurred or post-processed output? I mean, Was the output blurry even after applying undistortion algorithms?

Thanks in advance for your answers!