Livox-SDK/livox_horizon_loam #26: [laserOdometry-2] process has died [pid 26345, exit code -11]

Olive-Wang, 2020-12-08:
when running ' roslaunch loam_horizon loam_livox_horizon.launch ' with Lviox Horizon on Xavier NX, it gives the information as below, could anyone give me a favor?

> [laserOdometry-2] process has died [pid 26345, exit code -11, cmd /home/nvidia/workspace/drone_ws/devel/lib/loam_horizon/laserOdometry __name:=laserOdometry __log:=/home/nvidia/.ros/log/3f7580d0-3956-11eb-a88f-00e02def12ea/laserOdometry-2.log].
log file: /home/nvidia/.ros/log/3f7580d0-3956-11eb-a88f-00e02def12ea/laserOdometry-2*.log

may useful information
Xavier NX： ubuntu 18.04; ROS Melodic; ceres slover 1.14.0; pcl 1.8; Eigen 3.37

h-wata, 2021-02-02:
I faced same issue.

Olive-Wang, 2021-02-20:
> I faced same issue.
Have you fixed it ?

mick-h, 2021-03-16:
I have been getting the same issue.  It seems like it was initially caused by a lot of instances of `laserCloudCornerLast` being empty when the following call was made at `laserOdometry.cpp:521`:
`kdtreeCornerLast->setInputCloud(laserCloudCornerLast);`

This resulted in the error:
`[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!`
being thrown multiple times.  After this had happened quite a few times the `laserOdometry-3` process would die.

A great deal of these were caused by my rosbag having a lot of points recorded while the lidar was stationary.  I removed this data and started to get a proper point cloud being generated, but the same thing started to happen later in the bag.  As a stop-gap measure I placed a conditional around the two `setInputCloud` calls to stop them happening if the arguments were empty collections.  This allowed the processing to proceed further into the data, however there continued to be a lot of iterations where the `laserCloudCornerLast` was empty, and after a period the `laserOdometry-3` process would still die before getting very far into the bag.

It would be helpful if @Livox-SDK or someone else in the Livox team could explain what causes that collection to be empty, and whether or not it's possible to continue processing with some modifications to the algorithm.

Livox-SDK, 2021-04-25:
Hi all,
I suspecting that this error is caused by 'roslaunch loam_horizon loam_livox_horizon.launch' is not compatible with pcl 1.8. 
Would you mind if tryring to run this code with pcl 1.7?
@mick-h @h-wata @Olive-Wang

Rotoslider, 2021-04-27:
I have the same issue of it failing after a while.` [laserOdometry-3] process has died` RVIS just stops but the terminal window still chugs along like nothing is wrong. 
When do you plan on making this compatible with newer versions of PCL? All my other SLAM software is working with PLC 1.8. Maybe its an easy fix for you I hope.