Livox-SDK/livox_horizon_loam #32: How to determine "extrinsic quaternion" in External IMU rosbag?

congphase, 2021-05-26:
I intend to use IMU data from a NovAtel receiver. [Here](https://github.com/Livox-SDK/livox_horizon_loam#42-external-imu-rosbag) in your README, it determines the extrinsic quaternion as (0, 1, 0, 0) and modifies the launch file to those numbers.

The confusion is, as far as I know, quaternion keeps changing during the operation of the IMU as the vehicle moves. How can we determine it as fixed numbers like that?

Livox-SDK, 2021-06-04:
I think you may need to calibrate your external IMU Then replace those calibration parameters in loam_livox_horizon_ext_imu.launch
<rosparam param="ExtIL"> [0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0]</rosparam>

seajayshore, 2021-06-17:
Can you advise what the units of that calibration parameter [0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0] are??

And how would I actually calibrate the external IMU? Would the IMU manufacturer normally provide these tools with standard format?