Livox-SDK/livox_horizon_loam #33: loopclosure

huangsiyuan0717, 2021-07-15:
I see the mapping dont have loopclosure,  and i test the mapping compare use imu(horizon built-in imu) or not. But there is no obvious difference in the results.
Anyone meet the same problem?

ypLincode, 2021-08-05:
the IMU data seems unnormal, need to be smoothed?

ypLincode, 2021-08-05:
> I see the mapping dont have loopclosure, and i test the mapping compare use imu(horizon built-in imu) or not. But there is no obvious difference in the results.
> Anyone meet the same problem?

@huangsiyuan0717 Hi, I am using data also, the IMU data seems unnormal, need to be smoothed?

ypLincode, 2021-08-06:
> I see the mapping dont have loopclosure, and i test the mapping compare use imu(horizon built-in imu) or not. But there is no obvious difference in the results.
> Anyone meet the same problem?

@huangsiyuan0717 Hi, I am using the data also, the IMU data seems unnormal, need to be smoothed?